Livox-SDK/livox_ros_driver #164: LIVOX MID-360 not able to detect plzz help!!!

shettygr, 2023-12-11:


shettygr, 2023-12-11:
@Livox-SDK

EPVelasco, 2023-12-17:
Hi, I have installed the software that is on this [page](https://www.livoxtech.com/mid-360/downloads)

Also, I have created a docker file to run the LIVOX MID360 sensor with ROS noetic. You can see my [repository](https://github.com/EPVelasco/livox_ros_driver2) . In the README is how to create the docker and run it.

shettygr, 2023-12-19:
thank you , i will try these one and get back to you

shettygr, 2023-12-27:
hey @EPVelasco can the livox viewer2 installed on the raspberry pi4 with ubuntu20.04 64 bit version

EPVelasco, 2023-12-27:
The [manual](https://terra-1-g.djicdn.com/65c028cd298f4669a7f0e40e50ba1131/Mid360/Livox_Viewer_2_User_Manual_EN_v1.2.pdf)  specifies the requirement to ensure that the CPU or PC intended for Livox usage supports AVX (Advanced Vector Extensions 2). In addition, Intel users are recommended to use CPUs with 8th generation Intel® Core™ i5 processors or higher.

Note that the [repository](https://github.com/EPVelasco/livox_ros_driver2) provided is compatible exclusively with AMD architecture. Raspberry Pi, being based on ARM architecture, needs a different installation procedure not configured within the repository's Docker.

shettygr, 2023-12-27:
@EPVelasco iam going to place the livox mid-360 on drone,  so there should be some companion computer  on the drone right so iam using raspberrypi for that i need to install the livox viewer2 on rpi4 and its done by ubuntu 20.04 os 

But the raspberry pi uses the ARM architecture , but the ubuntu version in my laptop is AMD 20.04 os 
so iam not able to launch LIVOXviewer.sh in raspberrypi because rpi is ARM and the error is like "exec format error"

EPVelasco, 2023-12-27:
I'm not sure if livox viewer2 will work on a RasperryPi. 
In my experience, I used the [LiVOX repository](https://github.com/Livox-SDK/Livox-SDK) to install a LivoxMid-40 on ARM computer (Jetson Nano), which was going as an on-board computer for a drone. But I used ROS noetic to be able to visualize the point clouds.

shettygr, 2023-12-28:
@EPVelasco  can you share me some images of the point cloud data , so that i will use ros2 and get the visualization from mid-360 lidar 
https://github.com/EPVelasco/livox_ros_driver2 is the best one to get the point cloud data from mid-360

shettygr, 2023-12-28:
@EPVelasco if i use the above repository can ii get the point cloud data using docker image ..... 
and iam using the raspberry pi5 of 8gb and it is ARM architecture

EPVelasco, 2023-12-28:
Hi, I am not working with ROS2 right now.
The above repository was tested on AMD architecture. I don't have right now any Docker image that has been tested with any ARM architecture cpu.

I am working on an image that works for Raspberry pi. If it works, I will comment it in this issue.

shettygr, 2023-12-29:
Thank you i will also try these thing ,and if get any querry i will let you know

shettygr, 2024-01-05:
HI. @EPVelasco can u please share me the video of the process from scratch so that i can do the same because iam getting the errors in sdk and iam not able to figure out how its  is gonna works please can you do it for me so that i can implement these on my drone and here is my email (shettygr07@gmail.com)
i badly need these to be done by next week

shettygr, 2024-01-08:
so hi, @EPVelasco  iam getting these output after execting the livox sdk2 
[image]
what does these mean

EPVelasco, 2024-01-08:
Hi, apparently you have communication with the LIVOX mid 360. Have you already installed the [livox_ros_driver2](https://github.com/Livox-SDK/livox_ros_driver2). 
I have sent you an email with the title Livox Mid 360 with Raspberry pi, please check your email inbox

shettygr, 2024-01-08:
yes i have already installed the ros2 drivers and i trying to get the point cloud data like x,y,z and i am triying these in my desktop(ubuntu 22.04) not in raspberrypi

EPVelasco, 2024-01-08:
Personally I have not tested the LiVOX Mid-360 sensor in ROS2. I know it works but I don't have any environment with ubuntu 22.04 right now to test. Have you tried to launch the instructions ["For ROS2"](https://github.com/Livox-SDK/livox_ros_driver2?tab=readme-ov-file#for-ros2)

shettygr, 2024-01-08:
no ,not yet iam triying with livox sdk as of now and iam getting that as an output

shettygr, 2024-01-11:
@EPVelasco  hi iam able to run the launch the ros2 and rviz is also getting open but the visualization of the livox mid-360 in rviz is not happening so it is issuse with the code or what

JackQuests, 2024-03-19:
Hello! Could you please tell me how you have physically connected the Livox Mid-360 LiDAR to the Raspberry Pi? In my case, I am using the Raspberry Pi 4B. What cables have you used?

shettygr, 2024-03-19:
@JackQuests the cable which comes with the livox mid-360 three in one aviation cable ,so i connected to LANport on the raspberrypi

JackQuests, 2024-03-29:
thanks!! @shettygr

Puranjit, 2024-06-05:
> iam using livox mid-360 and it's not able to detect in the livox viewer and i am using ubuntu 20.04

Make sure that your Private and public firewall is turned off, this usually leads to not generation of any data.

JackQuests, 2024-07-17:
Hi, I've been working extensively to get the Livox MID-360 Lidar to work with Ubuntu 20.04 on my Raspberry Pi 4, but I haven't had any success so far. I've followed various steps mentioned in this thread, but I'm still encountering issues.

Could anyone kindly share detailed steps or tips on how they managed to get the Lidar functioning on Ubuntu 20.04? Any help would be greatly appreciated as I'm running out of ideas and really need to get this up and running soon.

Thanks!