Livox-SDK/livox_ros_driver #166: ubuntu18.04 mid-360 ROS,not error,not data out,plzz,help!!!!!!!!!!!!

DardNight, 2023-12-27:
~$ roslaunch livox_ros_driver livox_lidar_rviz.launch
... logging to /home/hg/.ros/log/74ed7982-a480-11ee-8672-08bfb8a54859/roslaunch-hg-16551.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hg:41339/

SUMMARY
========

PARAMETERS
 * /cmdline_file_path: livox_test.lvx
 * /cmdline_str: 100000000000000
 * /data_src: 0
 * /enable_imu_bag: True
 * /enable_lidar_bag: True
 * /frame_id: livox_frame
 * /multi_topic: 0
 * /output_data_type: 0
 * /publish_freq: 10.0
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /user_config_path: /home/hg/ws_livox...
 * /xfer_format: 0

NODES
  /
    livox_lidar_publisher (livox_ros_driver/livox_ros_driver_node)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[livox_lidar_publisher-1]: started with pid [16570]
process[rviz-2]: started with pid [16571]
[ INFO] [1703659788.848835629]: Livox Ros Driver Version: 2.6.0
[ INFO] [1703659788.849838903]: Data Source is raw lidar.
[ INFO] [1703659788.849936738]: Config file : /home/hg/ws_livox/src/livox_ros_driver/config/livox_lidar_config.json
Commandline input bd:100000000000000
Invalid bd:100000000000000!
Livox SDK version 2.3.0
broadcast code[47MDLA900200281] : 1 0 0 0 1 0
Add Raw user config : 47MDLA900200281 
Parse timesync config fail
Disable auto connect mode!
List all broadcast code in whiltelist:
47MDLA900200281
Livox-SDK init success!
[ INFO] [1703659788.850189373]: Init lds lidar success!

DardNight, 2024-01-03:
yep!!!!!!!!!!!!!

cjc021022, 2024-11-09:
> yep!!!!!!!!!!!!!
你好，我在Ubuntu18.04设备上没有error但是也没有点云输出，提示我“No tf data. Actual error: Fixed Frame [livox_frame] does not exist”。而在我的Ubuntu20.04设备是上是正常的，请问这个问题您当时是怎么解决的呢？

Hello, I have no error but no point cloud output on the Ubuntu 18.04 device. Prompt me "No tf data. Actual error: Fixed Frame [livex_frame] does not exist". My Ubuntu 20.04 device is normal. How did you solve this problem?